calcpsu/cphp82001 #3: Alternative battery connection

grob6000, 2024-09-16:
For others building who don't want to use the tiny JST-SH connector for the battery, should add a small wire connection for battery.

MilovdZee, 2024-09-17:
And why bother with a connector? My battery had a different connector anyway so I just soldered the wires directly on the board. Larger pads would indeed be nice.

grob6000, 2024-09-24:
I included the connector as I prefer to be able to add/remove the battery for testing.
* I've connected the battery +/- to the mechanical pads on the connector and made them a bit larger again, for doing what you've done. Great suggestion. No through-holes, as I don't want the B+ to be able to be contacted from the back. See 2951f1485d85d3fa7d55ea8e0cb5fbce0e2f9556.
* If there is no temperature sensor on the battery, then either populate R13 10k (battery normal charge with 10h shutdown, this is safest and what I like to build), or omit R13 - leaving TS floating (i.e. omit R13) or high (you can short pin2 and B+ on the connector) enables TTDM which disables 10h timeout. Refer to [datasheet](https://www.ti.com/lit/ds/symlink/bq21040.pdf). Note added to schematic: 7d6d5fe98b52145ec4c7ec134e7c0ce91a6d1320

closed 2024-09-24 by grob6000

MilovdZee, 2024-09-24:
I misread VSS as being VCC. But VSS is GND and so the 10k resister is perfect already. Thanks