calcpsu/cphp82001 #4: R1 value in schematics 1k8 but in BOM 1k3

MilovdZee, 2024-09-25:
ssia

grob6000, 2024-09-26:
The BOM was outdated; I've re-run direct from KICAD and it should now match the v4 schematic (0dc3c41eb4f7892238718d715c050627dc70e223). Thanks for spotting.

R1=1.8k provides USB charging current of 540/1.8 = 300mA, which is an ideal safe charge current of 0.2C for a 1500mAh battery
R1=1.3k would be 415mA, which is close to 0.2C for a 2000mAh battery.

R1+R8 sets the charge current when connected to the HP original charger (in-system). This is more the critical value. I know the datasheet says 10.8k max (50mA min), but hear me out:

The original HP charger supplies 50mA constant current (mine does 49.5mA pretty consistently, I'd be keen to collect more data but given the simple 2 transistor design I'm surprised it's better than +/-2mA).

When the 50mA supply is fed to the pads, the voltage will rise to above B+, and U3 will switch off, preventing this from directly feeding B+ - great. Note that U7 (NCP380) does not prevent backfeed (acts like a diode), so when the pad voltage exceeds 4.4V, U4 will switch on, and connect the pads to VIN of the U1 (BQ21040) via Q2 and U7 (which might be bypassed, if using the regulated design). U1 then starts up and charges the battery at ISET.

* If the total current consumed exceeds what the supply provides (>50mA), then the supply voltage will drop. While U1 will throttle current if the voltage drops below 4.3V (can maintain down at least as far as 3.5V), U4 will switch off at VOUTF=4.4V; prior to any throttling occurring. U4 has a 240ms reset timer, so the effect is a 240ms oscillation (flashing red CHG light). With U7 present, the ~0.3V body diode drop will help a little, but 
* If the total current consumed is less than the supply current (<50mA), then the charging works as normal, and the input voltage is regulated by the 5.6V zener.

I did consider relying on the BQ20140 current throttling (i.e. by replacing Q2 with a diode), however this means the charging IC is powered even when the battery is idle. In this case, it will not meet the power good threshold, and stay in standby mode. However, standby current is 125uA. In comparison, U4 consumes 30uA only. The 4.4V threshold must be above the charging output voltage (4.28V max).

Therefore, it's kind of critical that ISET is low enough that the total current draw does not exceed the constant current input supply, such that a small margin is available to run through the 5.6V zener and maintain the pad voltage >4.4V. Using the table below, even 12.8k is marginal.

| R1 | R8 | R1+R8 | ISET <br> MIN <br> KISET=470 | ISET<br>Nominal<br>KISET=527 | ISET<br>Max<br>KISET=605 | Recommend? |
| -- | -- | -- | -- | -- | -- | -- |
| 1.8k | 10k | 11.8k | 🟢 39.8mA | 🟢 44.7mA | 🔴  51.3mA |  |
| 1.8k | 11k | 12.8k | 🟢 36.7mA | 🟢 41.2mA | 🟠 47.3mA | ✔️|
| 1.3k | 10k | 11.3k | 🟢 41.6mA | 🟠 46.6mA | 🔴 53.5mA | |
| 1.3k | 11k | 12.3k | 🟢 38.2mA | 🟢 42.8mA | 🟠 49.2mA | ✔️ |

Note the datasheet does provide KISET values for 25~60mA (and lower!):
[image]

Now you've made me look at it again, I could in theory put 1V of diode between Q2 pin 3 and VCC, such that the idle voltage of 5.6V meant 4.6V at VCC (plenty for a 50mA charge), and at the cutout of VOUTF=4.4V, VCC was 3.4V - which would let U1 current throttle as designed, and thus the ISET could be 50mA (so 9.1k would work), and charge would be 20% faster.

MilovdZee, 2024-09-27:
Wow, a very comprehensive answer :) I'll study it :)

grob6000, 2024-10-15:
R1 and R8 values revisited for v5: 30ffbc13f589585e5461ab38b471176c49de2603

closed 2024-10-15 by grob6000