calcpsu/cphp82001 #5: Insufficient current from HP power supply + low battery charge state = no charge / flashing CHG LED

grob6000, 2024-09-27:
Raised in https://github.com/calcpsu/cphp82001/issues/4#issuecomment-2375859775

If current consumption exceeds the constant-current supply, input voltage will drop until U4 (APX810-44SA) switches off (VOUTF=4.4V). This will result in U4 cycling VCC with its 240ms delay. This threshold (which must be above max B+ (4.3V) to ensure U1 is unpowered when not charging) is above the VCC where U1 (BQ21040) will current throttle (<4.3V). As such, to charge, we require current consumption to always be below input current (allowing some to bias D2 (5.6V zener), maintaining the input voltage >4.4V), and thus the current setting needs to be conservative: ISET is 36.7~47.3mA (with range of KISET from BQ21040 datasheet, assuming no tolerance on resistors). Even this is marginal, as at the upper limit, +2.3mA for U1 + CHG LED, we might exceed available supply current. Add to this the HP supply is nominally 50mA, but in practice has tolerance - mine is measured at 49.5mA, but others may vary more. Currently assuming tolerance of +/-2mA, i.e. minimim 48mA, to design to a safe margin, the ISET may even need to be reduced.

If we could allow U1 to throttle, we could consume maximum available current for charging, regardless of the supply tolerance. Also, BQ21040 datasheet indicates that in this mode, the 10h timer is slowed to 20h, which would enable a 1000mAh or so per charge instead of only 500mAh. As such, it may be advantageous to encourage this mode rather than avoid it (in backfeed charge).

Testing using the USB input, I've worked out that BQ21040 will throttle well down to VCC=3.5V, and charge a nearly full cell (B+=4.1V) at VCC=4.5V, ICHG > 50mA. Ideally then VCC should be below VOUT, such that the throttling range for U1 (3.5~4.3V) corresponds to the backfeed voltage mode range for VOUTF (4.4~5.6V). A diode here would provide the drop. While it would prevent load sharing with USB input, for this design the USB is not accessible in-system so this is not required. Should an in-system alternative input (e.g. QI charging?) be added, this would require an alternative VCC input which diode feeds both VCC and VOUT (and U3 will take care of disconnecting B+ from load).

As B+ can be as low as 3.1V, we should assume VCC might drop close to this (even though datasheet says 3.5V min). As such, to avoid cycling V(diode) should be 4.4V-3.1V = 1.3V. This means, for excess current available, VOUT=5.6V, VCC = 4.3V, which is a bit close to the minimum for charging a full battery - would prefer 4.5V as tested - therefore 1.1V. [BAS116](https://assets.nexperia.com/documents/data-sheet/BAS116GW.pdf) has 1.1V at 50mA, which would be ideal.

Plan is to to a test on V3.5 board, and if this works, V5 will be updated to include the diode.

crozone, 2024-10-04:
Potentially outside the scope of this issue, but would this change allow compatibility with the USB to HP charging cables that are available on eBay?

[image]

They are implemented with a ~0.5V drop Schottky diode to drop the USB 5V to 4.5V for the logic supply, and a 68 ohm current limiting resistor for the charging circuit to give roughly the 40mA battery charging current. Would this be enough to maintain charging on the lipo?

grob6000, 2024-10-04:
Hi, thanks for providing the info on those cables! Yes, currently 40mA limit on the supply would likely cause this problem.

I built a test version using a 1N4148 and 9k1 R1 and it was quite happy charging even with about 25mA available, so I'd expect it would work with these cables.

For anyone with rev3.5 or earlier, you can cut the VCC track at the top of the board (runs between pin3 of Q2 and cathode of the USB diode), and solder a 4148 in there.

Currently working on an updated design (unregulated battery output only, which is demonstrably the better option for nearly all cases, with the new diode), so hang in there a week or so and it will be ready to go.

grob6000, 2024-10-15:
Implemented with v5: 30ffbc13f589585e5461ab38b471176c49de2603

closed 2024-10-15 by grob6000