calcpsu/cphp82001 #9: Tight fit in a HP35

grob6000, 2024-10-28:
One tester has reported the assembly is a tight fit in their HP35 - difficulty engaging the cover latches.

Consider reducing as far as practicable the case thickness/dimensions for the 2000mAh battery (10.5mm thick), otherwise the standard design will be reducing to 1500mAh capacity (8mm thick).

crozone, 2024-11-06:
Related, do you have a recommended PCB thickness for manufacturing these?

grob6000, 2024-11-06:
I use 1.6mm standard boards, but feel free to experiment with thinner. A
touch less thickness on the board may help here!

Alternatively, can always use a 8mm thick battery (and a thinner case;
uploading the model on my todo list).

grob6000, 2024-11-07:
Now added the 8mm battery case option. This is significantly thinner (1.5mm less), albeit will be limited to a ~1500mAh 803450 battery: 137d82596de65c59421a0eff91bd49e9d54f383c