calcpsu/cphp82001 #10: Mouser or Digikey saved basket for BOM?

crozone, 2024-11-20:
I'm currently building out the mouser shopping card for the v5 BOM and getting some choice paralysis on the generic parts.

Do you happen to have an existing saved basket for mouser or digikey? If not, should I build one out and submit it for future use?

grob6000, 2024-11-26:
Hi there! BOM has been refereshed with e6dfacf37bfa06bdd88ff2dee844c53cb2cf4b45, apologies for the confusion.

Happy to receive a link for a successful cart, otherwise I'll likely clean up my last Digikey BOM and provide a link in the readme.

grob6000, 2024-11-26:
Digikey shared list here: https://www.digikey.com.au/en/mylists/list/Z6K09H5IB0

closed 2024-11-26 by grob6000

crozone, 2024-11-27:
Thanks, this is a massive help! I'm currently working on a Mouser version as well. There's a few substitutions that I'll need to double check.